westphae/stratux #19: Change "Cage" button label to "Level"

westphae, 2017-03-14:
'nuff said.  Maybe other options to "Level": Calibrate or Cal?  Reset?

Nokomis449, 2017-03-14:
"Reset to Level" removes all doubt as to its purpose.  I also suggest a "Please wait..." window (or a plane icon tumbling on the screen) since the calibration takes more than a few seconds.

westphae, 2017-03-14:
Thanks.  Good idea on the dialog box, I've had that thought myself; your comment confirms it.  I think I will have the "auto-level" done in the next release, so I'm thinking of changing the button to just "Reset AHRS".  That's what it really does--the leveling is (was) just one aspect of that.

westphae, 2017-03-25:
Closed with ae92bae0d3af5c914c173ee9adb12dd04aba5c05.

closed 2017-03-25 by westphae